AgonPlatform/agon-vdp #26: Capture bitmap from area of screen

stevesims, 2023-09-02:
It may be useful to be able to generate new bitmaps from an area of screen.  This may aid in some graphical effects.

The underlying vdp-gl does not support this _directly_, however it can be achieved by fetching individual pixels using getPixel

julianregel, 2023-09-05:
This was functionality added to the Acorn GXR ROM:

>7.7 Defining sprites from the screen
As well as using the editor to design a sprite it is also possible to ‘pick up’ any rectangular area of the screen and make this into a sprite. To do this you should mark the rectangle by moving to its bottom left-hand corner and then its top right-hand corner. Then use the command
*SGET n
or alternatively
VDU 23,27,1,n,0,0,0,0,0,0
Either of these will define sprite n to contain whatever is currently in the
rectangle. Then this sprite may be used in exactly the same way as any other.

We're already using VDU 23,27,1 in the Agon VDP for another purpose, but perhaps the same sort of approach could be taken?

stevesims, 2023-09-05:
lol - yeah `VDU 23,27` is being used for sprites, but command `1` is to upload an RGBA bitmap.

it feels like a misstep to follow the Acorn VDU system, and even to make use of the same initial VDU sequence for sprites but to then make the implementation incompatible.

to be honest, I'd kinda half forgotten about the existence of the GXR.  back in the day I only had a passing awareness of it, and I suspect the number of times I encountered it in the wild I could count on one hand.  :grin:  having said that, I did program for the Archimedes a _load_ and the graphics system in that included most, if not all, of the GXR functionality - altho I tended to use higher-level BASIC ops, and mostly wrote desktop apps...  if I ever did want to use sprite functionality on RISC OS, I'd use the OS_SpriteOp SYS call...

that incompatibility _really_ bugs me :grin:

julianregel, 2023-09-05:
Hmm, if compatibility with the MOS and RISC OS in terms of VDU commands is desirable, perhaps we could extend the "legacy modes" VDU option to provide backwards compatibility, not just with modes (as currently implemented), but also sprites (and anything else that is currently implemented, but improving would be a breaking change). 

This would effectively make the VDU 23,0,193 setting an "Agon VDP 1.03 compatible" mode?

Not sure how much effort that is and whether it would be worth it.

stevesims, 2023-09-05:
I'm not sure how desirable compatibility is either :grin:

my perspective on this is skewed tho.  I'm coming at this as a former Acorn Electron, BBC Micro, and Archimedes programmer, so every incompatibility I see as undesirable.  every incompatibility is a potential source of confusion.

taking more of a newbie perspective I'm not sure I'd be quite so bothered that this _new_ computer doesn't work the same as a 30-40 year old machine.  the fact that code out there written for the old machines doesn't _quite_ work can _maybe_ be addressed with notes about porting such old code.

it's still potential friction tho.  examples of code that was originally written for the old Acorn machines will almost all need adjustment before they'll work.

a more completely friction-free experience would be to be able to skip that adjustment step.  that would mean changing the mode definitions for (at least) modes 0-6 to match up with the modes on the Beeb, and implementing various graphics things that aren't there just yet, and adjusting the default sound system setup slightly.

this is essentially what the Archimedes did.  virtually all the old beeb stuff is there, and BASIC code tended to just work (the only exception that comes to mind is a lack of `ENVELOPE` support).

I think where I'm going with this is that we _could_ change the Console8 VDP to, by default, be more Acorn compatible.  We could use `VDU 23,0,193,1` to switch to "Quark 1.03 compatibility mode", and use `VDU 23,0,193,2` to be "Quark 1.04+ compatibility mode"

a related note to this is storable configuration.  the BBC Master and Archie supported `*configure` to set configuration values in battery-backed CMOS RAM.  I believe that both the ESP32 and the ez80 have some flash storage that could be used for this purpose.  being able to configure your Agon to start up in a particular compatibility mode could be useful.

julianregel, 2023-09-06:
A couple of thoughts following your last comment:

The VDU compatibility with Acorn would obviously cover the BBC MOS machines and Archimedes, but also presumably the current RISC OS 5 implementation running on the Pi. In that context, compatibility wouldn't be exclusively for 30-40 year old machines, but would also be compatibility with a current, actively developed OS. 

Again, not sure if this is actually something people would care about, but it's an angle I'd not previously considered: That the Agon platform isn't a million miles away from the RISC OS platform (especially if the BASIC V additions get merged in at some point).

Regarding modes 0-6, it was something I looked at for the reasons you specify. Unfortunately, the 256 line modes (i.e., all of them basically) don't work on modern monitors. One option would have been to scan double to 512 lines, but this wasn't in the VGA specification, so would have added potentially compatibility issues. There had been multiple reports of problems running the legacy mode 2 - 320x200 at 75Hz - (the main issue here being the refresh rate), so I tried to select modes that were very "safe" in fitting into the VGA specification.

Adding support for ```*configure``` (and presumably ```*status``` to view what's set) would be a nice addition!

stevesims, 2023-10-22:
Whilst I had noted here before that the current bitmap API makes use of the same VDU command sequence to upload a bitmap as Acorn had used to grab an area of screen, it is not actually true that this is incompatible.

Acorn's documentation says to use `VDU 23, 27, 1, n, 0, 0, 0, 0, 0, 0`.  The API we have implemented doesn't read/use that many bytes, does not include `n` as a parameter, and instead is in the form `VDU 23, 27, 1, width; height; data...`

We _can_ support Acorn's version by simply looking for a height of zero

We can even support Acorn's version of specifying the sprite/bitmap number as part of the command, as that is sent in the lower 8-bits of where the `width` value would be.  Potentially we can even also support full buffer IDs - although that could be _slightly_ confusing given how we have made the existing "select" command operate.

DjPoke, 2023-11-09:
Hi,
Grabing the screen (to send it to a bitmap) can be usefull to make GUI menus, grabing areas behind a menu before, and drawing on top of it after. I don't know how to do something like this for now.

stevesims, 2023-12-15:
implemented in #103

closed 2023-12-15 by stevesims