AgonPlatform/agon-vdp #28: Overscan areas, and adjustable pixels offsets for display?

stevesims, 2023-09-02:
Can the canvas be adjusted so that it has borders that are not actually displayed?

Often on 8-bit systems developers would adjust the video systems so that some edges of the screen were not displayed.  Those areas would exist, and could be written to, the video circuitry just wouldn't show anything.

Additionally on some systems it was possible to adjust the overall x and y position of the display.

These features are useful for things like smooth scrolling, as new areas of the screen could be drawn in advance of them being actually visible.

stevesims, 2023-09-02:
VGABaseController supports `moveScreen`, which moves by adjusting front/back porch.  That may cover the "adjust the overall x and y position of the display" side of this idea.  `shrinkScreen` appears as if it may address the other half of the ask here.

We'd therefore need VDU commands to place calls thru to these underlying API calls.

julianregel, 2023-09-05:
For my tile demo, I added a VDU command (VDU 23,0,196) which, when called, would draw a border around the edge of the screen. The parameters I implemented were:

Define the border colour:

```
VDU 23,0,196,0,r,g,b
```

Draw the border:

```
VDU 23,0,196,1
```

The border drawn is 8 pixels wide/high as it was designed to hide the flickering/jumping of the 8x8 tiles. However, I was thinking that the border size could be made configurable so as to allow the positioning of sprites > 8 pixels in size partially off-screen (e.g., if dealing with 16x16 sprites, you might want a 16 pixel border so as to allow display of the sprites partially off-screen).

This obviously has the side effect of reducing the visible screen size, but as I've been working at the resolution of 320x240, this resulted in a visible screen of 304x224, which is still significantly better that other 8bit systems (and I believe is the same as the Neo Geo, so not too shabby(!)).

By making the border drawing configurable by the programmer, it is possible to decide what (if any) graphics display over the border. The default/assumed use case would be to draw the tile layer(s), sprites, text etc. and then end the drawing loop with a call to the border to frame everything correctly, but it would be possible to add other graphical objects on top of the border if desired.

The other thing I saw when looking at the VDP code was that scrolling the canvas (vdu_sys_scroll()) would move everything on the screen, which wasn't desired when I was aiming to scroll a layer and not the other objects on the screen.

Not saying this is the best option, but it might be worth considering.

stevesims, 2024-03-21:
experiments around this were made in #124 

they weren't really particularly successful :grin:

the underlying VGA signal generation in vdp-gl needs adjusting to more properly implement this kind of thing.  that would potentially allow us to introduce a proper border system and allow for better per-pixel scrolling

stevesims, 2025-03-13:
as of VDP 2.12.0, vdp-gl's VGA signal generation is being performed in a slightly different way for all screen modes.  essentially all modes now work the same for scan-out as the 2, 4 and 16 colour modes, whereby each scan line for output is created on-the-fly from the underlying mode framebuffer.  (this change is what facilitates hardware sprites)

in principle, adjusting how these scan-out lines are prepared could allow for the creation of optional borders.  these would come at the cost of resolution, but would potentially allow for variable sized borders, and could also _potentially_ allow for a degree of hardware scrolling

it should be noted that horizontal hardware scrolling may have limitations - there may be performance issues when moving by numbers of pixels that don't equate to word sizes.  the performance impact would likely depend on the colour depth of the mode

additional support would be needed inside vdp-gl