AgonPlatform/agon-vdp #32: Implement more of Acorn's GXR VDU calls

stevesims, 2023-09-05:
Acorn's Graphical eXtension Rom (GXR) contained several more graphical primitives that we don't yet support.

It would be good to try to add some more of them.

This will be practically limited by the underlying features of FabGL - altho by the looks of it _most_ of the primitives are present.

(Thanks to @julianregel for reminding me of the existence of the GXR :grin: )

julianregel, 2023-09-05:
I think you will find this interesting: https://tobylobster.github.io/GXR-pages/gxr/index.html

:-)

stevesims, 2023-09-11:
Some further analysis...

_some_ of the GXR calls would be hard to implement on top of fab-gl, and arguably are not practical.  these are primarily the dot-dash line commands, and colour patterns.  the "logical inverse colour" plot variants are likely not doable - fab-gl _does_ support a `NOT` paint option, but that is only implemented for straight lines and non-filled rectangles.

the fill commands (flood fill, and "line fill left/right to (non-)background/foreground") are maybe just about possible, but would be very hard.  without delving deep into fab-gl and rewriting sections of it, those would likely involve pixel-by-pixel calls, and be inefficient.

some GXR calls are tricky, but potentially doable.  those include the solid line calls that omit first/last/both points.  implementing these would either involve re-calculating points so that pixels get omitted, or more likely grabbing the appropriate pixel(s) from screen before drawing and then re-drawing them after the line has been drawn.

some call should be relatively easy, as they are built into fab-gl.  examples would be filled circles, filled rectangles, copy rectangle.  The parallelogram fill should be doable as a fill path.  Circular arcs, segments, sectors and ellipses _may_ be possible but will need some investigation to understand their behaviour.

command variants of relative vs absolute coordinates should be doable for all commands.  similarly variants that choose to plot in current foreground/background should be easy.

stevesims, 2023-09-22:
As much of GXR as practical (with the exception of "sprite"/bitmap) calls were implemented in #43 

No support for `NOT` was included, however this _should_ be added in a future PR, as there is _partial_ support for that present in fab-gl for straight line plotting

stevesims, 2023-12-15:
support for horizontal line fill plot calls was added via #104 

to be made more efficient, a different implementation for those calls is needed inside vdp-gl.

the horizontal line fill calls form the basis of Acorn's flood fill system.  we can look to implementing flood fill when the line-fill operations are supported inside vdp-gl

as per earlier analysis, pretty much everything else requires implementation inside vdp-gl.

stevesims, 2024-03-21:
dotted lines is implemented in #163 

arcs, segments and sectors are implemented in #173

stevesims, 2024-06-14:
bitmap plot code was added quite some time ago

what remains of GXR PLOT codes are the two flood fill, and two ellipse operations.

Acorn systems also supported pattern fills

stevesims, 2026-04-20:
finally added support for the flood fill and ellipse operations in #348 

what remains to be implemented from Acorn's graphics system are support for GCOL painting modes beyond the base 0-7 set...  that's the likes of "pattern fill"

rather than directly copying Acorn's pattern fill tho we might want to consider a bitmap fill...