AgonPlatform/agon-vdp #37: Consider reverting to FabGL for its UI elements

stevesims, 2023-09-06:
The only real significant difference between vdp-gl and fab-gl is the former has removed the UI components from the latter.

Having the ability to have a VDP-driven GUI on the Console8 could be a very useful thing.  It may help with providing a simple user interface for basic system management, and to help make the console usable without a keyboard attached (i.e. for it to be a "console").

We should therefore consider reverting to using FabGL, and finding ways of exposing an API for the UI elements it provides.

snakebyte69, 2023-09-30:
Create a new FabGL fork, call it AgonGL

stevesims, 2024-02-06:
as vdp-gl has now further drifted away from fab-gl by adding in new features this won't be happening

closed 2024-02-06 by stevesims