AgonPlatform/agon-vdp #71: Cache bitmaps in fast RAM

stevesims, 2023-10-25:
Currently all bitmaps are stored in SPIRAM.  This is great for memory usage, but not so great for performance.

Ideally we should have a way to cache bitmaps in fast RAM.

Even better would be to have the system automatically cache bitmaps when they are used.  This would probably need to be restricted to bitmaps below a certain size (in bytes) for practicality reasons.  Ideally the system should automatically release items from cache when more fast memory is required for other things, such as changing screen modes.

A simple mechanism for helping to manage this may be to just keep caching until we run out of memory, and flushing the cache before screen mode changes.

stevesims, 2023-10-25:
practically, we would need to keep a record of "cached" buffers.

to decide which items we should _keep_ in cache as the cache is used, we probably need to store a usage count.  that alone may not be sufficient however - an "age" may also be useful.

buffers used for active sprites should be given priority to stay in cache.

if an "adjust" operation is performed on a cached buffer then we need to ensure that when that item is removed from the cache, the buffer is still the modified version.

this likely means the cache needs to track that the buffer has been adjusted.

removing an item from the cache will need to check to see whether the cached item has been adjusted, and if so the adjusted contents written back to SPIRAM.

"advanced mode" of this could potentially periodically write back adjustments, and remove stale items from cache.

stevesims, 2023-11-09:
preliminary experiments show that this does not _appear_ to significantly improve performance

julianregel, 2023-11-29:
I've been doing some work over the last week where I've pulled bitmaps from SPIRAM for display and similarly thought about having a cache for speeding things up. I similarly discovered that there was no noticeable improvement to doing this in internal RAM (at least, not significant enough to go to the effort of writing such a cache routine).

closed 2025-01-29 by stevesims