AgonPlatform/agon-vdp #80: Support for displaying characters 1-31

stevesims, 2023-11-01:
Some font codepages use characters 1-31, which the VDP treats strictly as control characters and thus are not possible to be displayed.

It would be useful to provide a mechanism for displaying such characters, and having them defined in a font.

A suggested mechanism to display them is to use some kind of control character to indicate the _next_ character should be displayed.

The logical VDU code to use here is 27, which is "unused" by Acorn, and defined in Richard Russell's later BBC BASICs to mean "send next character to the screen".

Allowing the appearance of these characters to be re-defined would also be useful, which currently is also not possible as the VDU sequence to re-define characters is `VDU 23,<charNumber>` and numbers less than 32 are control operations for the VDP system.  Most of those control numbers are _currently_ unused on Agon, but many _do_ have potential meanings if we follow implementations either on Acorn's OSs or Richard Russell's BBC BASICs.  Whilst we can use character 27 as an escape code to cause a character within the 1-31 range to be displayed, we are using `VDU 23,27` already for bitmap/sprite operations, so a different mechanism needs to be adopted for redefining characters in the range 1-31.

lennart-benschop, 2023-11-15:
Don't forget to add VDU 27,127 to be able to display character 127, which is currently not displayable either. Its bitmap can be changed though with VDU 23.

stevesims, 2023-12-15:
to allow for characters in the range of 1-31 to be redefined we could add in a new `VDU 23, 0` command.  this would simply replicate the existing "redefine character" logic

whilst we're at it, adding in a command to reset user defined characters back to their original definitions would be good to add

closed 2023-12-19 by stevesims