AgonPlatform/agon-vdp #96: Programmable Function Keys

eightbitswide, 2023-11-15:
Would it be possible to have definable function keys which could be configured in the autoexec.txt file at boot up?

Sample configuration:

F1="SAVE \""
F2="LOAD \"
F3="*BYE<CR>"
F4="AED "
F5="LOAD BBCBASIC.BIN<CR>"
F6="RUN<CR>"
F7="LIST<BR>"

lennart-benschop, 2023-11-15:
Should this be a MOS command instead, like *KEY on the old BBC Micro? Of course the VDP must also be able to pass the Function keys to MOS.

HeathenUK, 2023-11-15:
Although I can see why you *might* add it to VDP (for example making it easier to encode multiple-lines as in the above example), I agree it should probably be in MOS. I've put together a simple PR for this (and one other) bit of functionality [here](https://github.com/AgonConsole8/agon-mos/pull/29).

stevesims, 2023-11-16:
As noted already, this is functionality that is better placed in MOS than the VDP codebase.

This was already raised as an issue there as AgonConsole8/agon-mos#8 

Closing.

closed 2023-11-16 by stevesims