AgonPlatform/agon-vdp #121: Variable width font support?

stevesims, 2023-12-22:
Building on #120 it is possible using vdp-gl to use variable width fonts.  We could, therefore, potentially allow their use inside agon-vdp.

NB this would be an advanced feature, and _may_ not be entirely in keeping with agon-vdp functionality.  Since the text cursor system in agon-vdp works on a character position, rather than a screen position, and thus relies on fixed width fonts adding this would be difficult and problematic. Were it added, it may need to use a different text cursor system.

lennart-benschop, 2023-12-23:
BBC/Agon VDU does support the graphics cursor to write text (VDU 5(, so this might be adaptable for use with variable width fonts.

stevesims, 2023-12-24:
yeah - nice idea.  so a possibility here could be that we only support a variable width font via the graphics cursor

the problematic parts are areas that use character-based positioning.  for instance VDU 31, which equates to BASIC's `TAB` statement, VDU 28 that sets a text viewport, VDU 23,0,&82 that sends the cursor position back to MOS, and VDU 23,0,&83,x;y; which reads the character on-screen at the given text position

(it's curious that VDU 23,0,&83 is using 16-bit values for the text position - that feels like a mistake.  perhaps the VDP should adjust the behaviour of this command when VDU 5 has been set to instead accept screen coordinates?)

stevesims, 2024-04-03:
currently the underlying text rendering in vdp-gl only has partial support for rendering text in variable width fonts.  code to handle variable width fonts is only present in the `drawText`, `drawTextWithEllipsis` and `textExtent` canvas API calls

currently all our text rendering is done using `drawChar` which does not respect character widths - indeed presented with a variable width font, unless the font data has been carefully organised to ensure that all characters are aligned to a standard width, it looks like this call may fail - or at least will draw more than is needed (including a background block for the full width the the font is defined as having)

(it may be the case that characters _need_ to align to standard width)

stevesims, 2024-06-14:
`canvas->drawChar` can be fairly easily updated to support variable width fonts.

alternatively agon-vdp could use `canvas->drawGlyph` to draw characters rather than `drawChar` - _essentially_ however this amounts to the same thing, so it is probably better to enhance `drawChar`