AgonPlatform/agon-vdp #125: HEXLOAD Patch

TomSchimana, 2023-12-26:
It appears to me that the HEXLOAD patch is included in Quark 1.04, but not in any vdp version for Console8.

HeathenUK, 2023-12-26:
What makes you think that? The code is in there and hexload works fine on C8...

stevesims, 2023-12-26:
Yeah - hexload support is present in the Console8 codebase

I regularly check for additions to the Quark codebase and regularly merge the Quark main branch into the Console8 codebase. There is nothing present in Quark VDP 1.04 that wasn’t in the Console8 VDP 2.0.0 release.

closed 2023-12-26 by stevesims

TomSchimana, 2023-12-26:
As soon as `send.py` initiates the transmission, whether HEXBIN is active or not, a reset or restart is triggered. This occurs regardless of whether the jumper is set, as described. I'll now test it with the Quark firmware. However, with the current firmware from C8, a reset is immediately executed. Hence, I assumed that the patch is not included.

HeathenUK, 2023-12-26:
Are you definitely using the latest send.py from the repo? I also occasionally have issues with hexload causing a reset if Ive recently used e.g.PIO or a bidirectional terminal on the port, which is actually down to an underlying reset circuit in the ESP32 rather than the firmware itself, but goes away after a reboot (of the host).

TomSchimana, 2023-12-27:
I've finally managed to prevent any further resets (reinstalled the vdp, reuploaded all files, etc.). Thus, we can consider this issue resolved. However, it's still not working properly. Hexload indicates a transfer (dots and one an X?), but it never returns to the prompt. But I believe this is no longer a firmware-related problem. Thank you for your help.

HeathenUK, 2023-12-27:
The FIFO in the ez80 can hold on to bytes sometimes and spit them out when hexloading, but that doesn't result in hexloading hanging...

Have you definitely got the latest (October, from memory) send.py on your host and latest hexload.bin in /mos/ ?

TomSchimana, 2023-12-28:
I can't explain why it's working now. I switched to Quark VDP, but that didn't help. Just then, I updated both MOS and VDP back to the latest C8. Now it's running smoothly. Thanks for the assistance.