AgonPlatform/agon-vdp #187: Feature Request: Add command to restore original cursor functionality.

eightbitswide, 2024-03-29:
When formatting text, it seems strange to have a cursor that is a different color from the text. 

Requesting an additional VDU 23, 1, 5 to restore the behavior of the cursor prior to 2.7.0

stevesims, 2024-03-29:
the underlying cause for the cursor appearing in a different colour is a fix that has been applied to ensure that the cursor will always be visible

the old cursor drawing system used a "swapFGBG" draw operation, which swapped "foreground" and "background" colour pixels around.  this would fail when the cursor was  placed on top of a solid background that is neither the current text foreground or background colour

the new cursor drawing system instead plots the cursor in the current text foreground colour using an XOR plotting operation.  this ensures that the cursor will always be visible, no matter what colour pixels it is drawn on top of.

the down-side of this new drawing system is that it means if the text background colour is changed and the screen cleared to that colour then that is not born in mind when drawing the cursor, so the cursor will appear in a different colour to either the text foreground or background.  the cursor would only appear in the text foreground colour when placed on top of colour 0 (usually black)

an enhancement to the new cursor drawing system is to draw the cursor twice using two XOR plots - one using the current text background colour, and the second with the foreground colour.  this means that the cursor will appear in the current text foreground colour when placed on top of the current text background colour.  the cursor will still get shown in different colours when other coloured pixels are covered by it, but the behaviour will in most/more cases be perceived as "correct"

stevesims, 2024-03-30:
enhanced cursor plotting using a double-XOR plot is present in #189

stevesims, 2024-05-30:
another approach for cursor drawing would be to capture the area of the screen where the cursor is being drawn to a "native" bitmap, and restore that area when the cursor is hidden

for now the double-XOR solution appears to work

stevesims, 2025-01-29:
Closing - the double-XOR approach seems to work well enough

closed 2025-01-29 by stevesims