AgonPlatform/agon-vdp #201: Alternative cursor rendering idea

stevesims, 2024-04-08:
Rather than the current XOR based cursor rendering, we could render the cursor in a similar manner to sprites - copy the background and then draw a rectangle when showing, and paste the background when hiding.  doing this the cursor will always be able to render, and will always be shown in the current text foreground colour.

going down this route we could also, potentially, add support for using a bitmap as a cursor

(part of the "problem" with the existing cursor rendering system is that the routine that draws the cursor neither knows or really cares whether it's showing or hiding the cursor.  we separately track a binary to indicate whether the cursor is showing or is hidden, but the drawing routine simply doesn't care)

lennart-benschop, 2024-04-28:
The cursor will then be completely invisible when it is over a rectangle that is completely in the foreground colour. For example a space that was drawn in reverse video.

stevesims, 2025-01-29:
Closing this - the current cursor rendering seems to be completely adequate and up to the task.  as pointed out this idea itself is flawed

closed 2025-01-29 by stevesims