AgonPlatform/agon-vdp #225: You cannot read `CTRL+L` without cleaning screen

nihirash, 2024-06-09:
If you press `CTRL+L` in any place - you'll cleaned screen anyway(even in games or demos). 

And there no difference is echo enabled or not.

stevesims, 2024-06-09:
this is intended behaviour.  pressing CTRL together with L is equivalent to performing a `VDU 12` command which will clear the screen, as L is the 12th letter.  the same behaviour works for _most_ other letters that match up to a single-byte VDU command - initially it worked with CTRL+N and CTRL+O only to enable and disable "paged" mode, but this has been extended

to disable this functionality see you need to call [`VDU 28, 0, 98, 0`](https://agonconsole8.github.io/agon-docs/vdp/System-Commands/#vdu-23-0-98-n-turn-control-keys-on-and-off)

nihirash, 2024-06-09:
Got it! 
Thank you for explaining! 

It's not a bug!

closed 2024-06-09 by nihirash