AgonPlatform/agon-vdp #243: 3D VDU commands not registered with mdfs.net

rtrussell, 2024-06-30:
As far as I can see, the proposed 3D VDU commands haven't been registered at [Jonathan Harston's site](https://beebwiki.mdfs.net/VDU_23).  Jonathan coordinates extensions to VDU etc. to ensure that they don't result in any compatibility issues.

stevesims, 2024-07-09:
hi @rtrussell  - thanks for the comment

to be honest, I hadn't thought to register for an account and make edits to add in Agon's extensions to the VDU system.

in the currently released versions of the VDP, _most_ of Agon's extensions sit under VDU 23,0 with higher-value register numbers (over &80) as per the suggestion given on Jonathan's site.  this mechanism lets much of the Agon VDP system work, given the Agon's rather unique architecture - for instance, the audio system is managed using a whole mass of commands under VDU 23,0,&85.  as there are no similar extensions documented for other platforms it feels like it probably wouldn't be appropriate to add all of these commands to Jonathan's site - perhaps it might be appropriate to just add in a link to the Agon VDP documentation.

(of minor note here is that the Agon VDP's VDU 23 system does not use fixed-length commands - they are all variable length.)

there are only three extensions that the Agon has to the VDU command set that I can think of that sit _outside_ of VDU 23,0

firstly Agon has a more extensive set of commands for bitmaps/sprites - these sit under VDU 23,27.  unfortunately the original author of these commands did not _quite_ follow the way that Acorn's "sprite" API had worked - although that has now _mostly_ been resolved.  the extended commands would probably not be appropriate to add to Jonathan's documentation, although a link to the Agon documentation for them might make sense

secondly is support for a "fill path" PLOT command.  this is implemented in the plot code range of &D8-&DF.  this is a block documented as "reserved" - which felt like it was probably OK to use, since Acorn aren't going to be using it. 😁   I implemented this feature in such a way that felt to me to be sympathetic with the rest of the PLOT system, although it is not something I would expect to see implemented elsewhere

thirdly there is a feature called "hexload" that uses VDU 23,28 which is used to facilitate transferring programs from a computer attached to the VDP's USB serial port to the eZ80.  (in hindsight I would have placed that command under VDU 23,0, but that ship has sailed - this feature was integrated before these console8 VDP releases.)

as of right now the 3D VDU commands are very much in the experimental phase.  they currently are implemented with a set of commands under the Agon's "buffered commands API" (VDU 23,0,&A0), and have not appeared in any official release.  the commands for performing 3d drawing probably should be moved to a different command block under VDU 23,0 if/when it becomes part of a release.  some might argue that this feature could use VDU 23,29 - but as it relies on Agon-specific features that probably isn't appropriate

the "buffered commands API" is a unique feature of the Agon VDP system - it provides a mechanism for storing and managing blocks of data in memory attached to the VDP.  given the Agon's architecture where there is no shared memory between the VDP and the main CPU, and all commands/data has to be transferred to the VDP over a serial link, this is necessary for many features.  buffers are used on the VDP for storing all sorts of data, such as stored sequences of VDU commands, bitmaps, sound samples, and fonts.  the experimental 3d graphics system uses buffers for its models.  this approach isn't necessary on other platforms

stevesims, 2024-07-10:
Whilst I appreciate that strictly speaking variable length VDU 23 commands are not allowed, that ship has sailed.  This was a decision taken long before I got involved in the project, and has been a reality of the Agon platform for at least the past 18 months.

Fixing VDU 23 would result in almost all non-trivial Agon programs breaking.

Even if this somehow could be fixed, doing so within the constraints would result in serious performance degradation. The Agon only has a practical link speed to its VDP of under 750kbit/second, so every byte counts. (This is under 1500 bytes per frame.) Adding in extra zero bytes to pad out short commands, and extra bytes for packetising longer commands would have a major impact - possibly as much as 25%, depending on the application. Some of the more advanced Agon programs we have wouldn’t be practical, or in some cases even work, with that penalty.

If I were to release firmware that made the Agon’s VDU 23 work “correctly” hardly anyone would install it, and it would be the death knell of this firmware branch.

I can’t really see this incompatibility as a big issue. It just means that programs written for the Agon that use these VDU commands can’t be run on other versions of BBC BASIC. They can’t be run anyway and I would not expect any other version of BBC BASIC to want to implement any of Agon’s rather esoteric VDU 23 command set. There are better, easier, and more effective ways of providing equivalent functionality on other platforms, and a significant proportion of Agon’s commands just wouldn’t be relevant.

As for the new PLOT codes, I did check beebwiki and the documentation for your versions of BASIC before picking the codes I used. Currently the “implementation“ list on beebwiki does not list any PLOT codes beyond those in RISC OS.

I wasn’t aware that there was a way to request plot codes as this is not mentioned on the beebwiki site for VDU 25. Could you please let me know what the procedure is to do this?

rtrussell, 2024-07-11:
> Whilst I appreciate that strictly speaking variable length VDU 23 commands are not allowed, that ship has sailed.

Nobody can force you to comply with pre-existing standards, but unless your VDU stream is fully compatible with the long-established conventions there is no point in it having any compatibility at all.  You might as well start completely from scratch with a completely different encoding.

After all, within your 'variable length' VDU 23 sequences you are effectively embedding completely non-standard VDU commands anyway, because the 11th byte in each sequence is the start of a new command.  If by that point you're sending 'binary' data it could be anything.

> Even if this somehow could be fixed, doing so within the constraints would result in serious performance
> degradation.

Splitting variable-length commands into 10-byte 'packets', each of which is a 'valid' VDU 23 command, only 'wastes' two bytes out of every 10.  For example you could request that VDU 23,20 and VDU 23,21 - which are currently unused codes - be allocated to Agon as 'first packet of a group' and 'second or subsequent packet of a group' respectively.  Then each packet has a 'payload' of 8 data bytes, an overhead of 25%.

> I would not expect any other version of BBC BASIC to want to implement any of Agon’s rather esoteric
> VDU 23 command set.

On the contrary, because my versions of BBC BASIC do fully support 3D rendering (Direct3D in the case of _BBC BASIC for Windows_ and OpenGL in the case of _BBC BASIC for SDL 2.0_) I was indeed going to investigate the possibility of providing some compatibility with the Agon.

There is a degree of elegance in encoding 3D commands into the VDU stream, both from a performance perspective (it makes sharing the processing load between multiple cores or CPUs easier) and when mixing 3D rendering with, for example, textual output which I currently can't do at all.

> I wasn’t aware that there was a way to request plot codes as this is not mentioned on the beebwiki site for VDU 25.
> Could you please let me know what the procedure is to do this?

It's a **Wiki**: you can _edit_, as well as read, its contents!  So the procedure is to register with the Wiki so you can make the changes you propose (preferably in draft form, unless they're trivial like fixing a typo).  If you're unsure of exactly what change you want to propose you can use the **Talk** pages to discuss it.

stevesims, 2025-07-16:
closing this.

at this stage it's unlikely that the Pingo-based experimental 3D API on the Agon will become part of the Agon Platform VDP firmware.  if the VDP ever does get some kind of 3D graphics API, it will be built on top of the existing buffered commands API and the matrix manipulation APIs that contains.

> Nobody can force you to comply with pre-existing standards, but unless your VDU stream is fully compatible with the long-established conventions there is no point in it having any compatibility at all.

I couldn't disagree more

the Agon VDP has great compatibility both with Acorn's VDU 23 commands, and also implements some of the extensions introduced in BBC BASIC for Windows.  it'll happily ignore the padding bytes other platforms require.

it also has great compatibility with the rest of Acorn's VDU command set.

> You might as well start completely from scratch with a completely different encoding.

Dean's porting your version of BBC BASIC to the Agon was a very important part of him building out the Agon's firmware.  that influenced both the APIs offered by MOS, and it also made a lot of sense to have the VDP implement a VDU command interpreter.

the hardware architecture of the Agon has a 1mbit serial link between the CPU and VDP.  the "VDP protocol" that Dean implemented is very simplistic - from the CPU to the VDP it is a raw VDU command stream, with no encoding overhead at all.  there's no "data packets", just the raw data.  comms from the VDP to the CPU _do_ use data packets.

the main advantage of this approach was that it was pretty easy, simple, and fast to implement.

a "disadvantage" is the hardware architecture dictates that a load of new VDU commands are needed.  for example, beyond VDU 7 there's no "sound" VDU commands, but as the VDP does the Agon's sound output to implement BASIC's `SOUND` statement you'll need new VDU sound commands

I don't know for certain as I've not had the opportunity yet to meet Dean and talk about it, but I believe that the VDU 23 "mistake" came from a simple misunderstanding of how those commands worked on the BBC Micro and Archimedes - he just didn't realise they were supposed to be fixed length.

as I explained before, fixing the Agon to be compatible is a non-starter for many reasons.

> Splitting variable-length commands into 10-byte 'packets', each of which is a 'valid' VDU 23 command, only 'wastes' two bytes out of every 10. For example you could request that VDU 23,20 and VDU [...]

this analysis isn't correct, and the true overhead is much higher.

a 4 byte command requires 6 bytes of padding; an 11 byte command expands to 20 bytes.  this is a much higher cost than a 25% overhead, which is a best-case scenario that only applies when the "payload" size is a multiple of 8 bytes long and/or for "bulk data transfer" operations.  a sophisticated Agon program will use lots of smaller VDU 23 commands with much higher overhead costs.

the 1mbit link already places significant limitations on VDU command throughput, and thus on potential program performance...  indeed if the Agon VDP _did_ have a compliant VDU 23 implementation there would be a very strong argument to implement an "efficiency" mode to remove the need to pad VDU 23 commands out to 10 bytes, as that could provide a significant performance boost

closed 2025-07-16 by stevesims