AgonPlatform/agon-vdp #254: Changing text viewport breaks enabling the cursor

avysk, 2024-09-20:
After the following sequence

1. Disable the text cursor.
2. Change text viewport.
3. Enable text cursor.

the text cursor is not shown until something is printed on the screen. To demonstrate the behaviour try the following BASIC program:

```basic
❯ cat D:\Downloads\fab-agon-emulator-v0.9.60-windows-x64\sdcard\BUG.BAS
   10 VDU 22, 0
   20 VDU 23, 1, 0
   30 VDU 28, 1, 11, 70, 10
   40 PRINT("PRESS ANY KEY TO ENABLE CURSOR")
   50 A$=GET$
   60 VDU 23, 1, 1
   70 VDU 26
   80 A$ = GET$
   90 PRINT("FOO")
  100 A$ = GET$
```

Running it produces the following result:

line 10: the screen is switched to mode 0 and cleared
line 20: the text cursor is disabled
line 30: the text viewport is changed
line 40: the message is printed (no cursor, as expected)
line 50: wait for keypress
line 60: the text cursor is enabled
line 70: the text viewport is reset (**no cursor, why?**)
line 80: wait for keypress
line 90: the message is printed (the cursor is back, as expected)
line 100: wait for keypress

Notice that removing line 30 fixes the problem. The cursor is shown right after it is enabled.

avysk, 2024-09-20:
I probably should mention that I discovered this in an emulator, but I see the same behaviour on real AgonLight 2 from OLIMEX. Both are running Console8 VDP 2.10.1

stevesims, 2024-10-15:
Verified this behaviour, and it is indeed a bug.

Running _essentially_ the same program on an Acorn Archimedes emulator (with fixes for chosen screen mode and padding the VDU 23 commands) results in the expected behaviour, which is for the cursor to start flashing after the first keypress, before the second keypress to get "FOO" to show

on investigation, this is caused by the fact that the clipping rectangle for drawing operations wasn't getting reset when the viewport was changed.  so when you reset the viewports, the clipping rectangle was still for the restricted text viewport, and thus the cursor, which now sits outside of that rectangle, wouldn't be drawn.  printing something causes the clipping rectangle to be updated

I have a fix - a single line to ensure that the clipping rectangle gets reset when the "reset viewports" command is sent.  I'll raise a PR with that fix shortly, and it will appear in the next VDP release

PR #257 (2025-01-14): reset clipping rect on viewport reset
fixes an issue whereby cursor could be invisible after a viewport reset

fixes #254
closed 2025-01-14 by stevesims