AgonPlatform/agon-vdp #322: getchar() seems busted in vdm_terminal_mode()

kristofer, 2025-04-30:
I suspect I am using it all wrong, but trying to get to the FabGL vt100 mode is proving awkward when I'm trying to understand `getchar()`

Maybe not the smallest program showing it, but this starts up, goes into term mode and hangs, with nothing from the keyboard ending up as result of calling `getchar()`. It just hangs after printing the "input> " string. `getchar` never returns and I have to kill the emulator to try again.

```c

#include <stdio.h>
#include <stdlib.h>
#include <stdbool.h>
#include <agon/vdp_vdu.h>
#include <agon/vdp_key.h>

static volatile SYSVAR *sv;

int main( void )
{
    sv = vdp_vdu_init();
    if (vdp_key_init() == -1)
		return 1;
    vdp_terminal_mode();

    //int i = 0;
    while (true){
        puts("input> ");

        int ch = getchar();
        printf("Character entered: %c %04x\n", ch, ch);

        if ( ch == '\x7f' ) { exit(EXIT_SUCCESS); }
        //i++; if (i > 8) break;
    }

    vdp_terminal_mode();

    puts("Done");
    return 0;

}
```

I'm using the latest emulator (although the same thing happens on the Olimex board), the latest `agondev` C compiler. Emulator VDP is 2.11.0, the HWs is 2.14.1.

kristofer, 2025-04-30:
and seeing the long set of messages in discord about terminal mode (in all of its capriciousness) I'm closing this and going back to make my project just use VDU calls.

stevesims, 2025-04-30:
Terminal mode is in a way kinda straightforward - but it is a pure terminal

The problem you experienced is most likely that the `getchar()` function is attempting to read a keypress from the Agon keyboard - which works by the VDP sending a VDP protocol keyboard packet to MOS. That functionality is suspended when terminal mode is active.

As terminal mode is essentially a pure terminal, you need to read keyboard/input bytes from the VDP UART connection. I must admit that I don’t know _how_ you do that from a C program on the Agon, but I do know that’s what you need to do 😀